ecubus/EcuBus-Pro #1: win7 64bit第一次安装完成启动后出现A JavaScript error occurred in the main process

ryanbaw, 2020-06-11:
A JavaScript error occurred in the main process

Uncaught Exception:
Error: ENOENT: no such file or directory, open

frankie-zeng, 2020-06-11:
@ryanbaw Please provide your version information and use the newest version.

ryanbaw, 2020-06-12:
> @ryanbaw Please provide your version information and use the newest version.

version 0.1.13

Sudozh, 2021-12-21:
我也遇见了同样的问题，ECUBUS version v0.2.19, 电脑win11.
报如下错
[image]

frankie-zeng, 2021-12-21:
> 我也遇见了同样的问题，ECUBUS version v0.2.19, 电脑win11. 报如下错 [image]

删掉掉老的版本，重新安装一下v0.2.19 版本试试？自动升级的会有这个问题！

Sudozh, 2021-12-21:
还是不行，但是 v0.2.16 没有问题，可以安装不报错

frankie-zeng, 2021-12-21:
Using Release v0.2.20 !!!

closed 2022-01-04 by frankie-zeng