ecubus/EcuBus-Pro #2: DoIP UDS Send wrong Diagnostic Session Control type

ryanbaw, 2020-06-12:
the wireshark dump as below:

ECUBus send Request Diagnostic Session Control(0x10):
0000   00 e0 4c 68 01 d4 c8 e7 d8 ce bb dd 08 00 45 00
0010   00 36 0f d8 40 00 40 06 44 de c0 a8 32 b5 c0 a8
0020   32 06 b1 34 34 58 1b e4 3d a4 e8 76 5c ff 50 18
0030   40 23 61 71 00 00 02 fd 80 01 00 00 00 06 00 0d
0040   10 00 **10 81**

ECU response:
0000   c8 e7 d8 ce bb dd 00 e0 4c 68 01 d4 08 00 45 00
0010   00 35 b7 3b 40 00 40 06 9d 7b c0 a8 32 06 c0 a8
0020   32 b5 34 58 b1 34 e8 76 5c ff 1b e4 3d b2 50 18
0030   03 ec ab 1b 00 00 02 fd 80 03 00 00 00 05 10 00
0040   00 0d **03**

I use the ECUBus send Diagnostic Session Control(0x10) defaultSession(0x01), data should be 0x1001,but wireshark dump is 0x1081

frankie-zeng, 2020-06-12:
@ryanbaw please check whether you choose suppress. if the suppress is choosed, the highest bit7 will be 1 of subfunction.

frankie-zeng, 2020-06-12:
@ryanbaw by the way, can you install the app correctly?

ryanbaw, 2020-06-12:
> @ryanbaw please check whether you choose suppress. if the suppress is choosed, the highest bit7 will be 1 of subfunction.

yes, I choosed the suppress. the ECUBus send 0x1001 when i canceled suppress. Thanks

but there  are problems still, could you help me?
ECUBus send:
000   00 e0 4c 68 01 d4 c8 e7 d8 ce bb dd 08 00 45 00
0010   00 36 18 3e 40 00 40 06 3c 78 c0 a8 32 b5 c0 a8
0020   32 06 b9 0b 34 58 4e 81 32 80 f4 b1 27 2f 50 18
0030   40 23 5c 38 00 00 02 fd 80 01 00 00 00 06 00 0b
0040   10 00 10 01

UDS server response
0000   c8 e7 d8 ce bb dd 00 e0 4c 68 01 d4 08 00 45 00
0010   00 35 3f 08 40 00 40 06 15 af c0 a8 32 06 c0 a8
0020   32 b5 34 58 b9 0b f4 b1 27 2f 4e 81 32 8e 50 18
0030   03 ec a5 62 00 00 02 fd 80 03 00 00 00 05 10 00
0040   00 0b 03

the uds server respone NACK code: Unknown target address (0x03), anything wrong with my uds server?

ryanbaw, 2020-06-12:
> @ryanbaw by the way, can you install the app correctly?

the install error still exists. but does not affect use

frankie-zeng, 2020-06-12:
> yes, I choosed the suppress. the ECUBus send 0x1001 when i canceled suppress. Thanks
>
> but there are problems still, could you help me?
> ECUBus send:
> 000 00 e0 4c 68 01 d4 c8 e7 d8 ce bb dd 08 00 45 [...]

Maybe you can check the TA logic in here/ https://github.com/frankie-zeng/ECUBus/blob/master/src/uds/ipuds.js

frankie-zeng, 2020-06-13:
@ryanbaw Hi, I have setuped the doip server you used, If you doesn't change the server code, the server will show: NACK 03.
I change the build/install/doip-custom-simulation/EMS.properties physical address from 0x1001 to 0x1000. It can work. 
[image]

ryanbaw, 2020-06-16:
> @ryanbaw Hi, I have setuped the doip server you used, If you doesn't change the server code, the server will show: NACK 03.
> I change the build/install/doip-custom-simulation/EMS.properties physical address from 0x1001 to 0x1000. It can work.
> [image]

Great! It works. Thank you.

closed 2020-06-16 by ryanbaw