ecubus/EcuBus-Pro #11: UDS Over Ethernet does not support UDS 0x37 services

startqiu, 2021-12-22:
I found that UDS Over Ethernet does not support UDS 0x37 services, can it be supported?

frankie-zeng, 2021-12-23:
Hi, @startqiu UDS support insert 0x36, 0x37 in script.
It's not supported on GUI config page, because the general use cases need so many 0x36 services.

you can see a script example to learn how to insert 0x36,0x37
```
if(readData[0]!=0x74){
  return false;
}
/*calculate bs*/
var len = (readData[1] & 0xf0) >> 4;
if(len+2!=readData.length){
  return false;
}
var bs=0;
for (var i = 0; i < len; i++) {
  bs += readData[2 + i] * Math.pow(256, (len - 1 - i))
}
if(bs<=2){
  /*service number and blocknum*/
  return false;
}
bs-=2;

this.openFile(writeData[0]['Download File'].name,'r');

var payload=[];
var bn=1;
var size=parseInt(writeData[4].memorySize,16);
var data;
if(size>bs){
  data=this.readFile(bs);
  if(data.length!=bs){
    this.log("file no enough data");
    return false;
  }
  size-=bs;
}else{
  data=this.readFile(size);
  if(data.length!=size){
    this.log("file no enough data");
    return false;
  }
  size=0;
}
payload.push({
  type:"number",
  name:'blockNum',
  blockNum:bn,
});
payload.push({
  type:"input",
  name:'data',
  data:data,
});
var check=(writeData,readData)=>{
  if(writeData[0].blockNum!=readData[1]){
    return false;
  }
  if(size==0){
    this.insertItem(0x37,[]);
    this.closeFile()
  }else{
    if(size>bs){
      data=this.readFile(bs);
      if(data.length!=bs){
        this.log("file no enough data");
        return false;
      }
      size-=bs;
    }else{
      data=this.readFile(size);
      if(data.length!=size){
        this.log("file no enough data");
        return false;
      }
      size=0;
    }
    bn++;
    if(bn==256){
      bn=0;
    }
    payload=[]
    payload.push({
      type:"number",
      name:'blockNum',
      blockNum:bn,
    });
    payload.push({
      type:"input",
      name:'data',
      data:data,
    });
    this.insertItem(0x36,payload,check);
  }
  return true;
}
this.insertItem(0x36,payload,check);
return true;
```

startqiu, 2022-01-04:
ok,Thank you very much,I get it!

closed 2022-01-04 by startqiu