ecubus/EcuBus-Pro #21: 无法将函数定义与现有声明匹配

huangmingfeisss, 2022-03-27:
安装好node-gyp后，用msvs2015或msvs2017，npm install提示在此解决方案中一次生成一个项目。若要启用并行生成，请添加“/m”开关。多个错误，无法将函数定义与现有的声明匹配。
[image]
[image]
[image]
该如何解决，非常感谢。

frankie-zeng, 2022-03-28:
确保  "node-addon-api" 和 noge-gyp 安装正确，拿gyp 编译一个hello world试一下。
可以参考CI的脚步 [build](https://github.com/frankie-zeng/ECUBus/blob/master/.github/workflows/Build.yml)

hmsfeng, 2024-03-05:
> 确保 "node-addon-api" 和 noge-gyp 安装正确，拿gyp 编译一个hello world试一下。 可以参考CI的脚步 [build](https://github.com/frankie-zeng/ECUBus/blob/master/.github/workflows/Build.yml)

可以给个详细的编译步骤吗？各个工具对应的版本，谢谢了。

closed 2024-11-21 by frankie-zeng