ecubus/EcuBus-Pro #26: A few questions to ask

zzgg86, 2022-05-01:
There are a few others I would like to continue to ask:
1. Can you provide 27 security algorithm calculation examples and 31 checksum calculation examples, or save a GROUP containing the above usage?
2. Can the sub-functions of the security algorithm be added, or how to modify the sub-functions that are not in the list?
3. How to insert other non-diagnostic messages into the process?
thank you very much！

frankie-zeng, 2022-05-01:
[image]
[27test.zip](https://github.com/frankie-zeng/ECUBus/files/8598206/27test.zip)
you can use the newest version to open the project.

the example uses node crypto module to do some crypto calculations and put the result into the next service.

also, the second 27 service use a self define subfucntion 66(0x42).

zzgg86, 2022-05-02:
I'm sorry, I didn't understand your sample code very well. I'm not familiar with js programming. How do I get the seed for the following security algorithm and implement it programmatically. thanks！

void SA_GenerateKey(u8 * RndSeed,u8* key)
{
  unsigned char  MASK[4]={0x12,0x34,0x56,0x78} 
  unsigned char i;
  unsigned char caldata[4]={0};
  for(i=0;i<4;i++)
  {
    caldata[i]=RndSeed[i]^MASK[i];
  }

  key[0]=caldata[3];
  key[1]=caldata[2];
  key[2]=caldata[1];
  key[3]=caldata[0];

}

frankie-zeng, 2022-05-03:
[image]
[image]

zzgg86, 2022-05-03:
Thank you very much, the security algorithm problem has been solved. In addition, regarding the calculation of the checksum of the file, the file can be opened and read, but I don't know how to get the file size? Can you help me out?        
//eg：
this.openFile('FlashDriver.bin','r');
let ptr=this.readFile(1024);
this.info(ptr);

How can I get the size of FlashDriver.bin?

frankie-zeng, 2022-05-03:
var fs = require("fs"); // Load the filesystem module
var stats = fs.statSync("myfile.txt")
var fileSizeInBytes = stats.size;
// Convert the file size to megabytes (optional)
var fileSizeInMegabytes = fileSizeInBytes / (1024*1024);

frankie-zeng, 2022-05-05:
this.changeNextFrame("routineControlOptionRecord",crc0.toString(16));

zzgg86, 2022-05-06:
ECUBUS is very powerful and solves it perfectly. Thank you very much for your support, this software is still very useful. This topic can be closed. Thanks again for the support! In addition, how can physical addressing or functional addressing be configured in each service?

frankie-zeng, 2022-05-06:
[image]
thanks, you can create multi schedule and assign different addresses to each schedule.

zhuanwan223, 2022-05-11:
Hello，how does this interface start？

eddy-bitsensing, 2022-12-30:
Same last issue occured when I set physical/functional can id at the same time. Error message is THE PCAN Hardware is already being used by PCAN-Net. Frankie and zzgg86, Did you guys solve this problem?

JOSEPH129009, 2023-09-01:
> Thank you for your support, the checksum has been calculated, but now I calculate it in the preload of 31 services, and then modify the preload, but it seems that the checksum value is not sent out in the end, what is the problem? function preLoad(writeData) { .......... this.info(crc0); crc0=crc0^0xFFFFFFFF; this.info(crc0);
> 
> this.changeNextFrame("routineControlOptionRecord",crc0);
> 
> return true; }

Hi @eddy-bitsensing , 
May I ask, How do you write your crc function in the routine control 0x31 to automate the process?

closed 2024-11-21 by frankie-zeng