ecubus/EcuBus-Pro #49: UDS on LIN

Abor88, 2024-04-22:
Hello,
Is UDS on LIN supported? I tried sending UDS Default session and I see that Master Request is properly sent with ID = 60 (0x3C) but then I see Slave Response header sent with ID = 125 (0x7D) which is the PID and not the ID of Slave Response frame (I am expecting a frame sent with ID = 61 (0x3D)).
So in my case I see a timeout in Slave response but I have verified with CANoe that my slave is working in Slave response.

frankie-zeng, 2024-04-22:
Please setup correct NAD， Over LIN is support,  EcuBus as LIN master, will send 0x3c+ data, and 0x3d (header only)

Abor88, 2024-04-22:
Thank you for you response and your support.
My slave is responding on NAD = 6 so in ECU BUS tool I setup NAD with 
Send NAD = 6
Receive NAD = 0 (since it is not supported to set Send NAD = Receive NAD).
STmin and P2 set as default.
Am I missing something?

frankie-zeng, 2024-04-22:
makr CAN-OE work as diagnostic only mode.

Abor88, 2024-04-22:
I have tested my slave with CANoe separately.
Now in my setup I have only ECU BUS connected with a Peak PLIN device. No other tools are opened.

frankie-zeng, 2024-04-22:
sorry, I mean, you sould config CAN-OE LIN LDF schedule config as diagnostic only mode

Abor88, 2024-04-22:
Sorry, maybe I am not properly understanding.
I am using only ECU BUS now and no LDF is loaded.
I just opened the tool, configured Response NAD, create 1 UDS project and try to send default session.
What is missing in my setup?

frankie-zeng, 2024-04-22:
[image]

Abor88, 2024-04-22:
No ok, I can use CANoe.
The question is why should I use CANoe since I am using ECU Bus only?
Is it no possible to use only ECU BUS?

frankie-zeng, 2024-04-22:
try to capture the raw log from PLIN-View, 
[image]

Abor88, 2024-04-22:
[image]

frankie-zeng, 2024-04-22:
you slave doesn't response, check your slave code.

Abor88, 2024-04-22:
I can send also the screen of ECU BUS log in debug
[image]

The point is this: it seems that the slave is not responding but if I detach it from ECU Bus and on another PC I use CANoe, the slave is always responding

frankie-zeng, 2024-04-22:
anyway, you need check it in your slave side, why the slave doesn't response...,

closed 2024-11-21 by frankie-zeng