ecubus/EcuBus-Pro #79: [BUG] White Screen on Startup with Version 0.8.19 on Windows 11

ootonn, 2025-03-05:
### Description

I encountered a bug when using version 0.8.19 of EcuBus-Pro on Windows 11. Upon launching the application, it displays a white screen and does not proceed further, rendering it unusable. Below are the details to reproduce and investigate the issue.

### Environment

- EcuBus-Pro Version: 0.8.19
- OS Name: Microsoft Windows 11 Pro
- OS Version: 10.0.26100 N/A Build 26100
- Processor: Intel I7-12700
- RAM: 32GB

### Operation process

1. Download [EcuBus-Pro](https://github.com/ecubus/EcuBus-Pro/releases/download/v0.8.19/EcuBus-Pro.0.8.19.exe) from github assets.
2. Install to all users with administrator privileges.
3. Running, display white

### Supplement
Works fine with version 0.8.18 and below.

### Screenshots

[image]

frankie-zeng, 2025-03-05:
I reproduced the problem, i will fix it.

robingergg, 2025-03-05:
I reproduced the bug too. Thank you for the support.

frankie-zeng, 2025-03-05:
fixed in [0.8.20](https://github.com/ecubus/EcuBus-Pro/releases/tag/v0.8.20)

closed 2025-03-05 by frankie-zeng