ecubus/EcuBus-Pro #83: Vector support

claudiiolima, 2025-03-06:
Vector CAN Adapter support (VN1610, VN1630...)

frankie-zeng, 2025-03-07:
As you know, vector has CAN-OE, I don't know if it makes sense to support vector adapter

claudiiolima, 2025-03-07:
But it is possible to have the vector hardware without CANoe license.

frankie-zeng, 2025-03-07:
Got it, let me evaluate the workload, I need to buy VNxxxx to test it. 
`If it receives 5 upvotes we will move it to our backlog`

[image]

claudiiolima, 2025-03-07:
> Got it, let me evaluate the workload, I need to buy VNxxxx to test it. 
> `If it receives 5 upvotes we will move it to our backlog`
> 
> [image]

Maybe if you give me some guide how PCAN was implemented, I could try to implement and test Vector.

frankie-zeng, 2025-03-07:
@claudiiolima I appreciate it. I will write a doc to guide it. maybe you can setup the dev env firstly. [https://app.whyengineer.com/docs/dev/arch.html](https://app.whyengineer.com/docs/dev/arch.html)

frankie-zeng, 2025-03-07:
see [How To Develop New Adapter](https://app.whyengineer.com/docs/dev/adapter.html), If you encounter any issues, we can discuss them here.

felixvanoost, 2025-04-11:
I'm also interested in this feature! I can help test support on a Vector VN1640A.

frankie-zeng, 2025-04-12:
@felixvanoost Thanks for your interest! Welcome any PR.

frankie-zeng, 2025-05-16:
@felixvanoost @claudiiolima Hi, guys, [v0.8.36](https://github.com/ecubus/EcuBus-Pro/releases/tag/v0.8.36) has support Vector CAN Devices. Could you help to test it.

felixvanoost, 2025-05-16:
Thanks, this is great! I should be able to share some results in ~2 weeks.

frankie-zeng, 2025-09-08:
seems vector features are stable, close it.

closed 2025-09-08 by frankie-zeng