ecubus/EcuBus-Pro #84: Feature Recommendations - DoIP

robingergg, 2025-03-07:
When using the tool, it would be extremely handy to have the next options:

- Be able to select custom port: during development it may occur that the server's port is not the standard port - 13400.
- When flashing with DoIP, it would be really handy to be able to upload an image (.bin) file, as performing a real flashing by hand is impossible. Maybe I am missing something here, but this time I did not take the time to look into the code base as I did with IP.
- When creating a flashing sequence, it would be very helpful to upload the image file beforehand and it would automatically calculate the size of it so we don't have to manually set the memory size for routine control and for download request services.
- Also it would be great to get the maxBlockLengthNumber automatically from the response of the server.

I am aware that the user is probably able to so these from an automated TS script, although it would be cool to be able to have these features with the tool itself.

Thank you for your work!

frankie-zeng, 2025-03-11:
@robingergg I’ve added the [UDS-Group](https://app.whyengineer.com/docs/um/uds/buildInScript.html) feature in version `0.8.23`. By default, a pre-integrated group called [RequestDownloadBin](https://github.com/ecubus/EcuBus-Pro/tree/master/resources/buildInScript/RequestDownloadBin) is included.

[image]

I’d like to highlight that **users can now create, save, and reuse their own UDS-Groups in the future.**

robingergg, 2025-03-12:
I try to connect to server(stub) just like before with the next settings:

[image]

However I keep getting this error:

[image]

To me it seems like that the IP address should be fine as I am running my stub on local IP bind to 0.0.0.0 on port 13400.
First off, I only wanted to produce a simple communication like sending a soft reset (which I managed successfully with previous version). 
The logical address is also correct. 
When I am connecting with a simple python socket (TCP) it works fine.

frankie-zeng, 2025-03-12:
could you show me the Trace window log, seems the client can't find the server, (no UDP reponse).

robingergg, 2025-03-12:
Sure, here's the trace window

[image]

Although I guess it fails before it would catch this request, hence the IP

frankie-zeng, 2025-03-12:
your server should response the VIN request. Or you can use this way #82 , connect tcp directly.

[image]

robingergg, 2025-03-12:
Right, I choose the wrong option and I did not omit. Thanks.

robingergg, 2025-03-18:
I am here again, trying to use the app but unfortunately, I am receiving an error: **upper layer read timeout**:
[image]

I am only sending a ReadDataByID request with a valid DID (0xF190).
When I start the sequence I can see that the RoutingActivation request has been received and from server side the corresponding response is sent out:

[image]

After that I can see that the server receives the ReadDID request and sends correct answer as it supposed to:

[image]

[image]

The complete data that is being sent out is this:
`\x02\xfd\x80\x02\x00\x00\x00\x19\x0e\x01N \x00b\xf1\x901HGCM82633A123456`
If I am not mistaken, everything lines up. I tried to look into the code of EcuBus-Pro, to see in what case the error upper layer read timeout occur and I can see, that there is a pending response from the server, which the client cannot catch (in my current understanding).

Do you have a tip for this?

frankie-zeng, 2025-03-18:
welcome your questions, follow the spec, you need send ACK, then send the respone. 

[image]

also, share the trace window.

frankie-zeng, 2025-03-18:
\x02\xfd\x80\x02\ **x00\x00\x00\x19**\x0e\x01N \x00b\xf1\x901HGCM82633A123456
The address is correct?

closed 2025-04-02 by frankie-zeng