ecubus/EcuBus-Pro #87: Linux SocketCan Support

frankie-zeng, 2025-03-16:
Add linux socketCan support

megaweeman2, 2025-03-23:
would using the node.js socketcan extension https://github.com/sebi2k1/node-can speed up development? Or do you think creating a binding from scratch be a better solution?

frankie-zeng, 2025-03-23:
thanks, yes, there are many ways to do it.

> would using the node.js socketcan extension https://github.com/sebi2k1/node-can speed up development? Or do you think creating a binding from scratch be a better solution?

megaweeman2, 2025-03-29:
started trying to make a socketcan interface but ran into some issues. 

my code in src/main/docan/socketcan/socketcan.ts

i thought since socketcan was linux only making a swig wrapper for it is kinda pointless so i didn't add any other folders. thing i installed was the npm socketcan module.

```
import {
  CanAddr,
  CAN_ID_TYPE,
  CanMsgType,
  CAN_ERROR_ID,
  getLenByDlc,
  CanBaseInfo,
  CanDevice,
  CanMessage
} from '../../share/can'
import { EventEmitter } from 'events'
import { execSync } from 'child_process'
import { CanError } from '../../share/can'
import { CanLOG } from '../../log'
import { CanBase } from '../base'
import * as can from 'socketcan'

export class SOCKETCAN extends CanBase {
  event: EventEmitter
  info: CanBaseInfo
  channel: any
  closed = false
  cnt = 0
  log: CanLOG

  constructor(info: CanBaseInfo) {
    super()
    this.info = info
    this.event = new EventEmitter()
    this.log = new CanLOG('SOCKETCAN', info.name, this.event)

    try {
      this.channel = can.createRawChannel(info.id, true)
      this.channel.addListener('onMessage', this.callback.bind(this))
      this.channel.start()
    } catch (err) {
      throw new Error(`Failed to initialize SocketCAN: ${err}`)
    }
  }

  callback(frame: any) {
    const id = frame.id
    const data = Buffer.from(frame.data)
    const ts = Date.now() * 1000 // Convert to microseconds
    const msgType: CanMsgType = {
      idType: frame.ext ? CAN_ID_TYPE.EXTENDED : CAN_ID_TYPE.STANDARD,
      brs: false,
      canfd: frame.canfd || false,
      remote: frame.rtr || false
    }

    const message: CanMessage = {
      dir: 'IN',
      id,
      data,
      ts,
      msgType,
      device: this.info.name,
      database: this.info.database
    }

    this.log.canBase(message)
    this.event.emit(this.getReadBaseId(message.id, message.msgType), message)
  }

  setOption(cmd: string, val: any): void {
    // Not implemented
  }

  static override getValidDevices(): CanDevice[] {
    try {
      // Run `ip link` to find available CAN interfaces dynamically
      const output = execSync('ip link show', { encoding: 'utf8' })
      const devices: CanDevice[] = []

      const regex = /(\bcan\d+\b|\bvcan\d+\b)/g
      const matches = output.match(regex)
      if (matches) {
        matches.forEach((iface, index) => {
          devices.push({
            id: iface,
            handle: index,
            label: `Detected Interface: ${iface}`
          })
        })
      }

      return devices
    } catch (err) {
      console.error('Error detecting CAN devices:', err)
      return []
    }
  }

  static override getLibVersion(): string {
    return '1.0.0'
  }

  close() {
    this.closed = true
    this.log.close()
    if (this.channel) {
      this.channel.stop()
    }
    this.event.emit('close')
  }

  writeBase(
    id: number,
    msgType: CanMsgType,
    data: Buffer,
    extra?: { database?: string; name?: string }
  ) {
    let maxLen = msgType.canfd ? 64 : 8
    if (data.length > maxLen) {
      throw new CanError(CAN_ERROR_ID.CAN_PARAM_ERROR, msgType, data)
    }

    const cmdId = this.getReadBaseId(id, msgType)
    if (msgType.canfd) {
      maxLen = getLenByDlc(data.length, false)
      data = Buffer.concat([data, Buffer.alloc(maxLen - data.length).fill(0)])
    }

    return this._writeBase(id, msgType, cmdId, data, extra)
  }

  _writeBase(
    id: number,
    msgType: CanMsgType,
    cmdId: string,
    data: Buffer,
    extra?: { database?: string; name?: string }
  ) {
    return new Promise<number>((resolve, reject) => {
      const frame = {
        id,
        data,
        ext: msgType.idType === CAN_ID_TYPE.EXTENDED,
        rtr: msgType.remote,
        canfd: msgType.canfd
      }

      try {
        this.channel.send(frame)
        const ts = Date.now() * 1000
        this.log.canBase({
          dir: 'OUT',
          id,
          data,
          ts,
          msgType,
          device: this.info.name,
          database: extra?.database,
          name: extra?.name
        })
        resolve(ts)
      } catch (err) {
        const errorMessage = err instanceof Error ? err.message : 'Unknown error'
        reject(new CanError(CAN_ERROR_ID.CAN_INTERNAL_ERROR, msgType, data, errorMessage))
      }
    })
  }

  getReadBaseId(id: number, msgType: CanMsgType): string {
    return `${id}-${msgType.canfd ? msgType.brs : false}-${msgType.remote}-${msgType.canfd}-${msgType.idType}`
  }

  readBase(id: number, msgType: CanMsgType, timeout: number) {
    return new Promise<{ data: Buffer; ts: number }>((resolve, reject) => {
      const cmdId = this.getReadBaseId(id, msgType)
      const cnt = this.cnt
      this.cnt++

      const readCb = (val: any) => {
        clearTimeout(timer)
        if (val instanceof CanError) {
          reject(val)
        } else {
          resolve({ data: val.data, ts: val.ts })
        }
      }

      const timer = setTimeout(() => {
        this.event.off(cmdId, readCb)
        reject(new CanError(CAN_ERROR_ID.CAN_READ_TIMEOUT, msgType))
      }, timeout)

      this.event.once(cmdId, readCb)
    })
  }
}
```

not to familiar with electron-vite. do i need to reference it in any other files for the main program to update the available interfaces?

frankie-zeng, 2025-03-30:
@megaweeman2 I update the [doc](https://app.whyengineer.com/docs/dev/adapter.html#add-in-ui) about how to add in UI