ecubus/EcuBus-Pro #89: MacOS support

ruziev-dev, 2025-03-17:
Hi EcuBus dev team!

Could you add support your application for MacOS?
As I know Electron.js can be compiled for a MacOS and also there are some technical opportunity to support hardware on MacOS (PEAK, Lawicel)

frankie-zeng, 2025-03-18:
This feature request is now a candidate for our backlog. The community can to [upvote](https://app.whyengineer.com/docs/dev/feature.html) the issue. If it receives 5 upvotes we will move it to our backlog.

ruziev-dev, 2025-03-27:
@frankie-zeng We got 5 upvotes 😎

frankie-zeng, 2025-03-29:
@ruziev-dev Hi, it's basically ok, but I think I don't have enough budget to pay for Apple developer. You can fork this repository, compile and sign it yourself.

[image]

[image]

ruziev-dev, 2025-03-29:
@frankie-zeng I have tried clone repository and execute `npm i`, what should I do next?
All commands from npm scripts block are executed with errors

[image]

frankie-zeng, 2025-06-25:
@sengulhamza appreciate it. could you run the esbuild from terminal? 

[image]

sengulhamza, 2025-06-25:
@frankie-zeng after fix esbuild.exe to esbuild for Darwin yes I can.

[image]

frankie-zeng, 2025-06-25:
> I changed esbuild but got worker terminated error while building test script.

So, could you show more detail crash information?

sengulhamza, 2025-06-25:
> > I changed esbuild but got worker terminated error while building test script.
> 
> So, could you show more detail crash information?

@frankie-zeng 

After I clicked refresh test config button.
[image]

I clicked Run Test Config
[image]

frankie-zeng, 2025-06-25:
@sengulhamza see https://github.com/ecubus/EcuBus-Pro/pull/161