ecubus/EcuBus-Pro #93: J1939 support

megaweeman2, 2025-03-23:
Hello!

Thank you so much for making this project! One thing that might be useful would be J1939 support. such as with the DBC file and dynamic Source Address claiming.

repos listed below show some implementation examples. 

https://github.com/DanielMartensson/Open-SAE-J1939?tab=readme-ov-file
https://github.com/famez/J1939-Framework
https://github.com/juergenH87/python-can-j1939
https://docs.kernel.org/networking/j1939.html
https://www.npmjs.com/package/@csllc/j1939?activeTab=readme

frankie-zeng, 2025-03-23:
Thanks list the ref. Because this limited resource,
this feature request is now a candidate for our backlog. The community can to [upvote](https://app.whyengineer.com/docs/dev/feature.html) the issue. If it receives 5 upvotes we will move it to our backlog.

frankie-zeng, 2025-03-25:
@megaweeman2 If J1939 support is added, can you help test it, because I don't have a suitable ECU device for testing?

megaweeman2, 2025-03-25:
> [@megaweeman2](https://github.com/megaweeman2) If J1939 support is added, can you help test it, because I don't have a suitable ECU device for testing?

Ya! i got a couple ECU's and test rigs i can try with it.