ecubus/EcuBus-Pro #103: [BUG]Baudrate for PEAK PCAN-USB-Adapter can't be changed

Philippe707, 2025-04-04:
When selecting PEAK CAN as Hardware Device the "Frequency" parameter is not visible as it is for KVASER CAN.
For example for the  [PEAK PCAN-USB-Adapter](https://www.peak-system.com/PCAN-USB.199.0.html) the user can't change this parameter. As a consequence for this type of adapter only the default frequency value 500000 is selected behind the scene which results in a possible adapter configuration of 500k Baud only.
Making the frequency parameter visible for vendor PEAK would resolve this problem.

**Environment**
1. EcuBus-Pro Version: 0.8.27'
2. OS Name: Windows

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an adapter as mentioned above
2. Add a PEAK CAN device. Select a combination of CAN parameters which results in a valid baud rate i,e, 250k
3. Connect...Play button on main window.
4. Open PCANview and check the configured baud rate for the interface in use...it's always 500k

**Screenshots**
[image]
Frequency parameter visible for KVASER

[image]
No frequency parameter for PEAK

[image]
PEAK baud rate selection in code.

frankie-zeng, 2025-04-05:
@Philippe707 For PEAK， the Freq is here,

[image]

frankie-zeng, 2025-04-05:
@Philippe707 check the PR?

closed 2025-04-06 by frankie-zeng

Philippe707, 2025-04-06:
Did some first tests with PEAK USB CAN interface without connection to CAN network. 
Works on all permitted baud rates: 1000k, 800k, 500k, 250k, 125k, 100k.