ecubus/EcuBus-Pro #126: Example for read data by identifier

Philippe707, 2025-04-29:
Is there an example that shows the usage of the service ReadDataByIdentifier. I can't extract any response data in the script on positive or negative response.
Thanks

Philippe707, 2025-04-30:
Looks like ReadDataByIdentifier can not handle iso tp reply..payload more than 4 bytes?

frankie-zeng, 2025-04-30:
maybe you can share your trace data, and here to chagne the response data lenght:

[image]

Philippe707, 2025-04-30:
[image]

[image]

[image]

frankie-zeng, 2025-04-30:
I found the bug, I will fix it.

frankie-zeng, 2025-04-30:
@Philippe707 which version you are using?
these are my test in 0.8.34.

[image]

[image]

[image]

closed 2025-04-30 by Philippe707