ecubus/EcuBus-Pro #127: Not possible to define two individual routine control services both with same routine control type

Philippe707, 2025-04-30:
It is not possible to define two individual routine control services with individual routine identifier but both with same routine control type (i.e. routine start).

Each routine control type can only be used once...

[image]

frankie-zeng, 2025-04-30:
This is a function, not a bug, because I need to use data to find the corresponding service in the system. If the data is the same, I cannot find the corresponding service. 
But I can optimize this problem, for example, allowing the same routine control type but different routine ids, do you think this is necessary?

Philippe707, 2025-04-30:
Meanwhile i found a workaround with the script to change the parameters of the service and therefore us the service for individual routines.
It's just nice to be able to configure individual routine control services. I think your suggestion would makes sense. I can life with the workaround. Thanks

frankie-zeng, 2025-05-01:
yes, script can handle anything.

I'm posting this question here to see if anyone else has the same need.

frankie-zeng, 2025-05-20:
I will fix the issue in next version.

closed 2025-05-21 by frankie-zeng