ecubus/EcuBus-Pro #139: [求助] ZLG 配置UDS问题

Moxuan6, 2025-05-19:
版本：v0.8.35
设备：ZLG USB CANFD 200U

配置示例：
[image]

按照这样配置之后，一直是接收超时

[image]

Addr Format : 选择拓展帧，BRS 勾上 几种组合都试过了不行

frankie-zeng, 2025-05-19:
从Trace看确实是没有返回，还是要看看你MCU是否收到，然后是否返回

Moxuan6, 2025-05-20:
> 从Trace看确实是没有返回，还是要看看你MCU是否收到，然后是否返回

大哥，不好意思呀，昨天有点事，没看消息

今天我测试了下，昨天的配置，MCU能收到消息，上位机应该是没识别，我勾上 BRS 就正常了。

[image]

不过，我发现一个问题，如果 Trace 报文特别多的情况下，操作上位机的时候就会卡死，拔掉 ZLG 就好了，是不是ZLG驱动有点问题

frankie-zeng, 2025-05-20:
卡死 的情况能录个小视频吗

Moxuan6, 2025-05-20:
> 卡死 的情况能录个小视频吗

现在来看就是启动关闭的时候容易卡死
https://github.com/user-attachments/assets/28ebac33-5dbb-4ba9-b4f6-9ff65a355f61

frankie-zeng, 2025-05-20:
收到

closed 2025-05-22 by frankie-zeng